kr15h/ofxPiMapper #134: OS X error on exit

stephanschulz, 2018-03-27:
This add-on looks great. Thanks for making it.

I am testing it on OSX 10.12 with OF 0.10.
It works but on exit I get a "crash" which points to the MediaServer.h 

Any ideas how I can fix that?
I think it's something about not being on a PI.
[image]

kr15h, 2018-03-27:
I think it is related to threads and I think it will get away together with the directory watching thing. But for now, while it works...

stephanschulz, 2019-01-31:
FYI the problem still exists in OF 0.10.1 with fresh add-on download.