kr15h/ofxPiMapper #135: is it possible to disable MediaServer?

stephanschulz, 2018-03-27:
I'm planning to use your add-on to show a couple of live camera feeds. 
This means I will not need the MediaServer to constantly check the directories for new media.
Is there a way to disable the imageWatcher.beginWatch(); ?

thx

```
[ error ] ofDirectory: listDir:() source directory does not exist: ""/boot/ofxpimapper/sources/images/""
[ error ] ofDirectory: listDir:() source directory does not exist: ""/boot/ofxpimapper/sources/images/""
[ error ] ofDirectory: listDir:() source directory does not exist: ""/media/usb0/""
[ error ] ofDirectory: listDir:() source directory does not exist: ""/media/usb0/""
[ error ] ofDirectory: listDir:() source directory does not exist: ""/media/usb1/""
[ error ] ofDirectory: listDir:() source directory does not exist: ""/media/usb1/""
[ error ] ofDirectory: listDir:() source directory does not exist: ""/media/usb2/""
[ error ] ofDirectory: listDir:() source directory does not exist: ""/media/usb2/""
[ error ] ofDirectory: listDir:() source directory does not exist: ""/media/usb3/""
[ error ] ofDirectory: listDir:() source directory does not exist: ""/media/usb3/""
```

kr15h, 2018-03-27:
Yes, I plan to remove these checks since the watching was introduced with the slideshow source which I believe should stay as custom functionality.

stephanschulz, 2018-03-27:
agreed would be great to have a bool to toggle this option.
I forked your project. any advice where to properly place this toggle?

kr15h, 2018-03-27:
Have to dive into it... Hold on a bit.