kr15h/ofxPiMapper #136: Problem remote server

amsicora68, 2018-03-28:
Hello
I tried but it gives me this error:
/home/pi/openFrameworks/addons/ofxPiMapper/src/Sources/VideoSource.h:7:27: fatal error: ofxOMXPlayer.h: File o directory non esistente
  #include "ofxOMXPlayer.h"
the example of ofxPiMapper works fine instead.
Thanks
Marco

kr15h, 2018-03-28:
What system are you compiling on?

amsicora68, 2018-03-28:
raspberry 2 and 3 this error is on 2 with raspbian jessie

kr15h, 2018-03-28:
Did you clone ofxOMXPlayer into the openFrameworks/addons folder?

amsicora68, 2018-03-28:
Of course.The example of ofxPiMapper works fine.

kr15h, 2018-03-28:
Weird. Will have to check this on a Pi later.

magdesign, 2018-03-28:
Yup, can confirm on Raspian Jessie:
When trying to compile example_remote-server, following error terminates compiling:

```
/home/pi/openFrameworks/addons/ofxPiMapper/src/Sources/VideoSource.h:7:27: fatal error: ofxOMXPlayer.h: No such file or directory
  #include "ofxOMXPlayer.h"
```

pietrondo, 2018-03-28:
you need to add in addons.config
ofxOMXPlayer

pietrondo, 2018-03-28:
no sorry
in addons.make

kr15h, 2018-03-28:
Ok, fixed it 2c705c0a653a283b0a9ef0290fdb06484eaa65bf

closed 2018-03-28 by kr15h

amsicora68, 2018-03-28:
many thanks works well now!