kr15h/ofxPiMapper #137: access fboSource

stephanschulz, 2018-03-28:
Sorry for having so many questions. But I really hope to use the add-on for a new project.

How would I access the mapped fboSource?
I want to pass it in to a second mapping. 

   piMapper._application._mediaServer.fboSources[0]->drawFbo(); ?

But some elements are private. Before making them public I thought you might already have a way to access them.

kr15h, 2018-03-28:
With FBO sources it is that you initialise them in your code already. Thus you have full control over them.

kr15h, 2018-03-30:
Maybe I misunderstood. Do you mean you want to access the individual surfaces with texture mapping on them? That could be enabled in a way...

stephanschulz, 2018-03-30:
Here is an example.
top left you see how via openCv I warp a square source image back to looking square.
just below it you see the original skewed source.
on the right you see the piMapper adjusted, so that it looks good when projected.

ideally I can skip the opec warp and do it via piMapper. I guess do two passes. Pass the unwrap image to the second mapping.
[image]

kr15h, 2018-03-30:
I see. That looks like an interesting feature to have. Will look into this. I am slowly thinking of another way for the surfaces on general. Something that would allow me to change the type of the surface more easily. Will think about this.