kr15h/ofxPiMapper #138: Slide Show Source folder

TheodoreMichels, 2018-03-28:
Hi there - first of all, thanks so much for all of your work on this project.
Sorry if this has been covered elsewhere, but I was unable to find any documentation on how to use the "Slide Show Source"
Also, would it be possible to use the Slide Show Source to play videos in sequence as well?
If not, I'd be interested in helping if you can point me in the right direction.
Thanks in advance,
Teddy

magdesign, 2018-03-28:
Slideshow Source works only with images so far.
I takes all the images in the images folder and plays them.
Settings can be made in: /ofxPiMapper/example/bin/data/magslideshow_settings.xml
Just try it.

TheodoreMichels, 2018-03-28:
Hi @magdesign - thanks so much for the quick response.
I've selected the Slideshow Source, but I only see a black screen in the quad map.
I don't see any errors in the console, and everything else seems to be working - although I did see a message that reads "Radio List: Item with value Slide Show Source not found"

And thanks for pointing to the settings file - I wasn't aware of those options.

magdesign, 2018-03-30:
Maybe your images are too big in size?

TheodoreMichels, 2018-03-30:
I'm just using the images that come with the basic "example".
For what it's worth, I'm running the installation on Raspbian Jessie, not using the preconfigured image.

rychrd, 2018-08-09:
Hi @TheodoreMichels 

Did you resolve your issue? I think i am seeing something similar:

**[ error ] ofImage: loadImage(): couldn't load image from ""image1.jpg""
[ error ] ofImage: loadImage(): couldn't load image from ""image2.jpg""
[ error ] ofImage: loadImage(): couldn't load image from ""image3.jpg""
[ error ] ofImage: loadImage(): couldn't load image from ""image4.jpg""
[ error ] ofImage: loadImage(): couldn't load image from ""image5.jpg""**
**[ error ] magSlideShowSource::initialize: Failed to create slide show from folder /home/pi/openFrameworks/addons/ofxPiMapper/example_fbo-sources/bin/data/sources/images/**

I'd changed to an absolute path in settings

This is on stretch, with latest OF 0.10.0 and the default images

rychrd, 2018-08-09:
I don't think mine was the same issue, but i solved it by changing line 182 in magSlideShowSource.cpp from

**if (tempImage.load(file.getFileName())){**
to
**if (tempImage.load(file.getAbsolutePath())){**

TheodoreMichels, 2018-08-09:
Hi @rychrd - I haven't had a chance to revisit this since I posted it - I ended up going with a different solution.
But for what it's worth, I think you had a different issue - I wasn't getting those error messages.
Glad to hear you figured it out - thanks for the input.