kr15h/ofxPiMapper #139: Remote Client in Browser

magdesign, 2018-03-29:
Thank you so much for releasing the remote-client and remote-server, its just freakin' amazing!

In the next few months I will try together with my coder to create a client which will run in the webbrowser (same approach as you know from PocketVJ).
So the client will run also on the Rpi (as a webserver) and its commands will be sent trough a website.
I hope there is enough power on the RPi... 

My question is:
Can you give me some hints how I can talk to Remote Server (like pointing me to some files which do the talking) ?

kr15h, 2018-03-30:
I would build a simple Python-based server as a proxy. The Python script would establish a TCP socket connection to the oF app. The Python script would publish an API that would be accessible through something like `http://127.0.01:8080/opm`. There you could forward JSON packets from the Web Interface by using whatever you are using for AJAX.

magdesign, 2018-04-27:
Until I have the Webclient solution, 
I compiled the Remote clients for Linux and OSX:
https://pocketvj.com/downloads/MapperRemote.zip

Failing to build the Windows version in VisualStudio due this stupid backslash filepaths, it will not find the data folder and outputs an error. Using Windows is like learning an arabic language for me......just impossible.
If a windows geek reads this, please help so we can compile also the windows client.

kr15h, 2018-04-27:
It's cooking https://youtu.be/xv-UAn7Segs

kr15h, 2018-04-29:
I have AppVeyor set up to compile ofxPiMapper, it compiles fine with msys2 as well as Visual Studio 32 and 64 bit.

magdesign, 2018-04-30:
It compiles here too, but when you execute the .exe it crashes cause it does not find the data folder...

kr15h, 2018-04-30:
I see. I could start up my virtual machine at some point. Will take a look.

pietrondo, 2018-05-22:
good news...