kr15h/ofxPiMapper #140: Bugs in: example_remote-client

magdesign, 2018-03-29:
I figured out following bugs:

When creating a new surface (quad, circle or triangle) from the remote client, several things do not work right yet:
- The resize handle does not seem to work. Workaround = using +/- keys or close client and open in again.
- Moving the corners in triangle does not work. Workaround = close client and open in again.

And, colors of videos are rendered in green as soon as client connects. Workaround = restart mapper.

My test scenario:
- client: Ubuntu 17.10, of_0.9.8
- server: RPi3, jessie, of_0.9.8, 1920x1080, 492MB allocated memory (sudo vcdbg reloc)

hgshoggins, 2018-09-11:
I confirm the greenish rendering on the server as soon as the client connects.

pietrondo, 2018-12-05:
same problem.... greenish rendering

kr15h, 2018-12-05:
Could be because I am not using `ofPushStyle()` and `ofPopStyle()` when setting color for the green status connection dot. Will take a look.

kr15h, 2018-12-05:
Fixed 67609da29be628913a79997e85cace1718501529

closed 2018-12-05 by kr15h

pietrondo, 2018-12-06:
THX now It works!!