kr15h/ofxPiMapper #142: too slow when the resolution is 1920x1080

pietrondo, 2018-05-11:
Hi!
We have a little problem with the video when the output resolution is at 1920x1080: the video become very slow (we have tried a lot of codec and solution but nothing). The videos become slow!
Also with video at lower resolution and with the last raspberry 3b+.
Someone have some experience?
Thx!!!

kr15h, 2018-05-11:
You mean the video resolution is 1920x1080?

magdesign, 2018-05-11:
Make sure there is an audio track in the video, even if it is silent, this helps sometimes.
Using HD video often stutters in the first run, then it gets "smooth".
Are you using one 1920x1080 video or several videos ?
What compression ?
Fast SD card ?

pietrondo, 2018-05-11:
1)Thx very much for the help and the software
2)ok for the audio
3)the video are at 1280x800 resolution
4)only one video
5)samsung class 10
6)ffmpeg -i  -vcodec libx264 -profile:v baseline -preset:v slow -level 3.1
-refs 2 -coder 0

Pietro Capriata

http://www.pietrocapriata.me

CONFIDENTIALITY DISCLAIMER
Le informazioni contenute in questo messaggio di posta elettronica e negli
allegati sono riservate e confidenziali e ne sono vietate la lettura,
l'uso, la copia, la comunicazione e la diffusione in qualunque modo
eseguite. Qualora lei non fosse la persona destinataria del messaggio, la
invitiamo ad eliminarlo, dando gentilmente comunicazione al mittente
tramite e-mail di ritorno.

The information contained in this email message and any files transmitted
with it are confidential and privileged and any reading, processing,
distribution or copy of this material is strictly prohibited, in any form.
If you are not the intended recipient of this message, please immediately
delete it, giving the relevant communication to the sender by reply e-mail.

davidedezan, 2018-05-12:
Ciao Pietro riusciresti a spiegarmi brevemente come aggiungo dei video alla cartella source della sd con ofxpimapper installato? Una volta inseriti li dovrei riuscire ad aggiungerli alla schermata video per poi mapparli giusto?

pietrondo, 2018-05-13:
ciao... dimmi primaditutto:
hai un pc linux?

Pietro Capriata

http://www.pietrocapriata.me

CONFIDENTIALITY DISCLAIMER
Le informazioni contenute in questo messaggio di posta elettronica e negli
allegati sono riservate e confidenziali e ne sono vietate la lettura,
l'uso, la copia, la comunicazione e la diffusione in qualunque modo
eseguite. Qualora lei non fosse la persona destinataria del messaggio, la
invitiamo ad eliminarlo, dando gentilmente comunicazione al mittente
tramite e-mail di ritorno.

The information contained in this email message and any files transmitted
with it are confidential and privileged and any reading, processing,
distribution or copy of this material is strictly prohibited, in any form.
If you are not the intended recipient of this message, please immediately
delete it, giving the relevant communication to the sender by reply e-mail.

pietrondo, 2018-05-13:
> Ciao no, uso Windows

perché la soluzione più veloce è il caricamento da linux...
riesci a procurarti una macchina linux?

Pietro Capriata

http://www.pietrocapriata.me

CONFIDENTIALITY DISCLAIMER
Le informazioni contenute in questo messaggio di posta elettronica e negli
allegati sono riservate e confidenziali e ne sono vietate la lettura,
l'uso, la copia, la comunicazione e la diffusione in qualunque modo
eseguite. Qualora lei non fosse la persona destinataria del messaggio, la
invitiamo ad eliminarlo, dando gentilmente comunicazione al mittente
tramite e-mail di ritorno.

The information contained in this email message and any files transmitted
with it are confidential and privileged and any reading, processing,
distribution or copy of this material is strictly prohibited, in any form.
If you are not the intended recipient of this message, please immediately
delete it, giving the relevant communication to the sender by reply e-mail.

davidedezan, 2018-05-13:
Si non è un problema, ma cosa centra con l'immagine già flashata su sd messa nel raspberry?