kr15h/ofxPiMapper #143: How can I add video and images into folder

davidedezan, 2018-05-11:
Hi I'm new here. Can someone explain to me how can I add some video and images into the SD of the raspberry? Thank you!

kr15h, 2018-05-12:
One way is to copy your .mp4 file to USB flash device, insert it into the Raspberry Pi and reboot it. The video should show up in sources.

davidedezan, 2018-05-12:
Thank you very much for your reply. Sorry but this is the first time I approach with commands based on Linux.
So when I insert the USB drive with images and videos, how can I see it and how can I add videos on the display for mapping it?

kr15h, 2018-05-12:
Are you using the .img file from the repository?

davidedezan, 2018-05-12:
Yes

davidedezan, 2018-05-12:
Downloaded and flashed with etcher

kr15h, 2018-05-12:
It goes like this.
1. Encode your video using a tool such as the [Handbrake](https://handbrake.fr/), 720p recommended
2. Name it in a way you can recognise it later (e.g. video-with-zebras.mp4)
3. Copy it on a USB flash drive and insert it into the Raspberry Pi
4. Reboot the Raspberry Pi and wait until the mapping software comes up
5. Hit <4> on the keyboard to see the list of sources. "video-with-zebras.mp4" should be there.

davidedezan, 2018-05-14:
Hi I've done all the points, but in the sources there isn't my video on usb

magdesign, 2018-05-14:
Did you ever used this usb stick for emulating or installing an operating system?
If yes, you have to manually remove the hidden bootpartition on this stick.
If no, did you try it with formatting the stick as simple FAT format.
It might be that the image is not supporting all modern file formats like ExtFat etc.

@kr15h: might be missing on the image: sudo apt-get install usbmount dosfstools exfat-fuse exfat-utils

kr15h, 2018-05-14:
Ah yes, you are right, must have forgotten those packages. Will push a new version tonight.
Another way is to mount the SD card on a Linux machine (You can switch to Developer mode in Windows as far as I know) and put the videos in /home/pi/openFrameworks/addons/ofxPiMapper/example_basic/bin/data/sources/videos

But the new image file should be there soon.

kr15h, 2018-05-14:
Plus I think that usbmount is just enough.

magdesign, 2018-05-14:
something like: sudo mount /dev/sda1 /media/usb/

davidedezan, 2018-05-24:
Everything is working thank you! Now I have another problem, how can I add space on the partition to add videos? There is free space on sd card,but sudo say to me that there is no space...

marcelbueckner, 2018-06-17:
Hell everyone,
first of all thank you so much for this great tool!
I ran into a similar issue when I installed the image and tried to see the video or jpg files.
but they dont show up. Even after a reboot. USB Drive is formatted as fat32 (no fat option because of Win10)
how can I login to opm to get acess to the comand line to check if the raspi recognize the usb?
opm login: ...
Thanks in advance

phpanhey, 2018-12-05:
Have same issues like @LeeviLehtinen. Is there a fix now ? Any suggestions?