kr15h/ofxPiMapper #146: A problem with the presets system

olivercrush, 2018-05-23:
Hi,

I'm using the preset system but I'm running into a problem. I have actually 12 presets (so 12 <surfaces></surfaces> in my xml file) and I'm wondering if that's too much ? Because when at first when I switch there are no problems but after the sixth switch there's only white squares instead of my surfaces.

I don't think the problem comes from one of my preset since they all work, it all depends of the six firsts I switch. So if at launch I display the preset 1, 2, 3, 4, 5 then 6, every preset after that won't work. But if for instance I display first the preset 1, 3, 5, 7, 9, 11, it's all the others that won't work afterward.

Maybe do I have to clear the presets or something ? Also if I switch too quickly between presets, the program just stops working. I added a usleep method to try compensate that but it still happens sometimes.

I don't know if I'm very clear, sorry, english is not my native language. Anyway, thank you again for your work on this project !

kr15h, 2018-05-23:
Seems like you have a memory problem, it gets overloaded. Try to access RPi settings via `raspi-config` and change the GPU Memory to 256 or 512. Still it does not guarantee that you won't run out. The video unload is broken and that might fill up the memory. But try this.

olivercrush, 2018-05-23:
It works, I changed the GPU Memory from 256 and 512. Since I don't use videos in my presets, I don't have any problems. Thank you !

kr15h, 2018-05-23:
Osom! Happy to hear. Good luch with the project and let me know if there are more issues.

closed 2018-05-23 by kr15h