kr15h/ofxPiMapper #147: TouchBoard support

olivercrush, 2018-05-24:
Hi, it's me again.

I was wondering if there is a way to support the BareConductive's TouchBoard controller. I didn't see a feature about that in your code (that would be cool for the future tho) and I was wondering how you would incorporate that in the ofApp file.

Thanks a lot, and sorry if the question is too general

kr15h, 2018-05-24:
I think you should use [this tutorial](https://www.bareconductive.com/make/the-touch-board-as-an-hid-keyboard/) to make your touch board into a HID keyboard and then map the keys to specific functions of ofxPiMapper. Take a look at the [example_pocketvj ofApp.cpp](https://github.com/kr15h/ofxPiMapper/blob/master/example_pocketvj/src/ofApp.cpp) class to get an idea how to map keys to ofxPiMapper functions.

olivercrush, 2018-05-25:
Yes I thought of that, unfortunately I'm using the PiCap version so the tutorial for the original TouchBoard doesn't work. Sorry, I forgot the precision.. I'll look into a similar way to emulate the HID keyboard anyway tho.