kr15h/ofxPiMapper #148: Rotate a surface

olivercrush, 2018-06-13:
Hi,

It's not really a bug, but I have to map vertical hexagons and the hexagon surface you generate is an horizontal one. I repositionned all the vertices to match my shape, but it rotates the image I used. I will solve this problem by using an already rotated image, but I wanted to know if there is a way to simply rotate the surfaces (like the size system with the mouse in the ProjectionMapping mode). And if not, if you planned to add this feature ?

Thank you !

pietrondo, 2018-06-13:
good feature..

kr15h, 2018-06-14:
Yes, this would be a good additional feature. Currently I am a bit off with tasks, but later this summer I am planning of activating the development a bit.