kr15h/ofxPiMapper #149: ofxPiMapper on Raspberry Pi 3B+ - HELP: How to??

massimilianoC, 2018-06-15:
Hi, I'm trying to install oxfPiMapper on the new Raspberry Pi 3 B +... is it possible? I have problems with OpenFrameworks, gcc 6 and the latest Raspbian release. Any suggestion?
Thanks!

kr15h, 2018-06-18:
Download the latest available Raspbian Jessie here: http://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2017-07-05/

And then follow the openFrameworks guide on installing it on the Raspberry Pi (note that everywhere you see of_v0.9.8 you should replace it with of_v0.10.0 as it is the latest version). https://openframeworks.cc/setup/raspberrypi/raspberry-pi-getting-started/

Then clone ofxPiMapper (and dependencies) to compile the examples.

Another way is to download an image file from GitHub releases page, or you can also access it through ofxpimapper.com

kr15h, 2018-08-09:
I got a Pi 3 B+ and I see where is the issue. It does work with latest Raspbian Stretch only. Hold on, will upgrade the images. Otherwise you have to install things from scratch yourself which means

- Downloading and installing latest Stretch
- Downloading and compiling openFrameworks 0.10.0
- Compiling ofxPiMapper yourself

massimilianoC, 2018-08-10:
I solved installing the image on a Raspberry 3 B and then upgrading the bootloader for the 3B+,  so I can load the SD into a 3B+ and it's working really fine!

kr15h, 2018-08-10:
Can you share details on how you upgraded the bootloader?