kr15h/ofxPiMapper #153: Circle doesn't crop buffer

aubergene, 2018-12-12:
I'm trying to use the circle surface, and I want to layer it on top of other surfaces however it doesn't appear to crop the buffer correctly, it makes anything outside of the circle black, when it should be transparent. Is this a bug or just something I'm doing wrong? Thanks

[image]

kr15h, 2018-12-12:
Hi! The circle surface was a contribution and it is basically the same quad surface with a mask on it. I will take a look, maybe there is a way to make the black part transparent.

aubergene, 2018-12-12:
Thanks that would be great. Yes, that makes sense to just have a general way to mask. I'm pretty new to c++ and OF but tried this without luck, but I can continue to play around

```c++
fbo->getTexture().setAlphaMask(mask.getTexture());
```

c-mendoza, 2019-11-08:
Way late to this party, but I just submitted a PR that should fix this. FYI the CircleSurface was a total rush job and my intention was to keep working on it on a separate repo. Part of why it was done with a mask was to :

1. Not have to modify the texture mapping and perspective mapping widgets of the app.
2. Make mapping more straightforward for circles (if you have an ellipse shape that you are distorting you might not end up with an actual perspective warped circle).

So it was 100% a hack! Anyways, hope this helps someone.

magdesign, 2019-11-19:
@c-mendoza: Thanks for looking into this.
Unfortunately your fix does not build, see comment on pull request.

PR #166 (2020-03-15): Fix for CircleSurface mask
Makes the mask FBO have a proper alpha channel and uses blending functions to achieve the correct masking effect. Fixes #153
closed 2020-03-15 by kr15h