kr15h/ofxPiMapper #156: no fullscreen with Debian of Ubuntu

jgastonraoul, 2019-01-15:
Hi,

I've built the last version of ofxPiMapper (example_basic) with latest release of openFrameworks (0.10.1) on Debian 9.6 (with Openbox) and Ubuntu 18.04 (the default version with Gnome Shell I think) and I'm unable to go fullscreen, either with -f option or f shortcut...
I've already tried to comment #ifdef lines around ofSetFullscreen(true) in ofApp.cpp but it doesn't help.

My goal is to have it working on Debian on a Gigabyte Brix (Core i3).

Any idea ?

Regards,

Jérémie

kr15h, 2019-01-15:
Yes, that is normal as I removed the feature from examples. Try adding the following to the `keyPressed` method.

```
if(key == 'f'){
ofToggleFullscreen();
}
```

Hope it helps!

toughvj, 2019-01-27:
Change OF_WINDOW to OF_FULLSCREEN in main.cpp

magdesign, 2021-02-03:
Thanks, this workes:

Change `OF_WINDOW` to `OF_FULLSCREEN` in `/addons/ofxPiMapper/example_basic/src/main.cpp`

issue can be closed...

kr15h, 2021-02-03:
Allright!

closed 2021-02-03 by kr15h