kr15h/ofxPiMapper #157: Simple video slide show

toughvj, 2019-01-28:
Hi everybody. I've wrote a simple video slide show as an fbo source, that may be usefull for someone.
It's very basic for now.

It's getting files from sources/video folder, default plays first video in loop. Calling toggleSlideShow() we may switch to slide show and back to playing only current movie. There are also void playNextMovie() and playPrevMovie() which makes what it says.

In the future i would like to add some transitions between loaded movies.

[https://www.dropbox.com/sh/boflyyb5nkzh667/AAA3lu1xyw4K8ofMPt8NaOROa?dl=0](url)

magdesign, 2022-03-11:
this sound interesting to me, especially the playNextMovie() command.

However, your dropbox link is dead, would you mind putting this on github, so we can see what you changed and maybe add this as a merge to original ofxPiMapper.

toughvj, 2022-03-11:
this should work:
https://www.dropbox.com/s/h0v36j5nb7ha6gv/ofxPiMapper.zip?dl=0

magdesign, 2022-03-13:
Thank you very much!
I guess its in coded in the "mapOne" folder.
how do you do "playNextMovie()" ? is there a shortcut assigned or via dbus or is it not coded yet?

toughvj, 2022-03-14:
No problem, I abandoned this project some time ago but I'll try to help.
playNextMovie() is assigned to keyPressed event.