kr15h/ofxPiMapper #159: PiMapper -  openFramworks v0.10.1 - PocketVJ

slagteriet, 2019-03-27:
Hi Im running a build of the great pocketVJ project and wanted to add PiMapper to extend the greatness.

I got the latest version of openFrameworks running v0.10.1 combine with this install guide. [(link)](https://openframeworks.cc/setup/raspberrypi/raspberry-pi-getting-started/)

I then followed this GitHub instructions [link here](https://ofxpimapper.com/pages/github-instructions/)

But seems not to be working..  ?

> From the guide in link above..
> Checkout the stable version of ofxOMXPlayer.
> cd /home/pi/openFrameworks/addons/ofxOMXPlayer
> git checkout 0.9.0-compatible

kr15h, 2019-03-27:
Could be they are a bit outdated. Let me take a look... ; ]

kr15h, 2019-03-27:
```
cd /home/pi/${OFX_DIR}/addons/ofxOMXPlayer
./install_depends.sh

```

kr15h, 2019-03-27:
```
git checkout master
cd /home/pi/${OFX_DIR}/addons/ofxOMXPlayer
./install_depends.sh
```

Will update the instructions on website.

slagteriet, 2019-03-27:
Thank you @kr15h Will try it.

kr15h, 2019-03-27:
Then there is this CI file that I use to make automated builds: https://gitlab.com/kriwkrow/pimapper/blob/master/ci/rpi/pi-gen/stagePiMapper/00-install-opm/00-run.sh Maybe you find something useful there as well.

slagteriet, 2019-03-30:
@kr15h did it work ? :)

kr15h, 2019-03-30:
Well, it works with the automatic build. 
You can download the image file (artifacts) here https://gitlab.com/kriwkrow/pimapper/tags/v1.0.0
or from the ofxpimapper.com website https://ofxpimapper.com

kr15h, 2019-03-30:
I am building with the master branch of openFrameworks, and latest version of Raspbian

closed 2019-05-25 by kr15h