kr15h/ofxPiMapper #162: recognizing jpgs and mp4?

mmersereau, 2019-05-24:
I've seen questions about usb mount problems but I think this maybe another issue. I've been using ofxpiMapper for a few installations over the past few years. That said I've been using an .img from a couple years ago as well. I had a pi crash and had to reinstall pimapper again but the files do not show up, everything I think is rendered for best results. It also used to originally copy the file to the raspi from the usb after boot. I guess a couple questions, Is there anything particularly new in setting up videos now. Placing them in the data folder for videos and pics does not recognize them either, but the stock videos and images are there. ok best. thanks for your time.

kr15h, 2019-05-25:
Hi! Sorry to hear that. What has changed is that the image build process is automated now and the ofxPiMapper image is assembled using latest Rasbpian, openFrameworks and ofxPIMapper updates. There should be no problem using jpg and mp4 files, but it might be also because of other dependencies, such as ofxOMXPlayer, in the stack. I would like to have my own solid video playback solution instead of that, waiting to have enough free time to be able to build that. For now it would be good if you could share the images and I could use them to test a bit as it is hard to tell.

One thing that you could try is to not remove the USB stick as the current way is that everything is deleted from the folder except the demo video if you remove the stick. And.. This actually might be the problem! Let me look into it. And thanks for letting me know!

kr15h, 2019-05-25:
Yes. Actually that might be the case. The videos are deleted on startup. I will figure out a quick solution. The new image should be available tomorrow.

kr15h, 2019-05-25:
Here we go https://gitlab.com/kriwkrow/pimapper/-/jobs/219321797/artifacts/raw/PiMapper_v1.0.1.zip
Let me know if there are more issues.

closed 2019-05-25 by kr15h

mmersereau, 2019-05-25:
OK great! I'll give it a shot tomorrow and let you know.

mmersereau, 2019-05-26:
Works like a charm. :) Thank you so much. This is one of the best creations
for the pi.
best
Michael