kr15h/ofxPiMapper #169: Raspberry Pi 4 B

techgunk, 2020-01-16:
I'm trying to get this working on some Rpi 4 B to use at a gallery, is the Rpi 4 not supported?

kr15h, 2020-01-16:
Not yet

—-
Krisjanis Rijnieks
Creative coding. Projection mapping. Solutions
https://rijnieks.com

eshkrab, 2020-04-29:
Hello! 

Considering using this on a project and I'd love to use RPI4, happy to help with the porting effort if I can. 

Would you mind briefly summarising what would be necessary for the port if you know? OF has been ported to RPI4 now, so is there a lot that needs to be done on top of that?

magdesign, 2020-04-29:
@eshkrab 
Why don't you test it and report what is not working.....

eshkrab, 2020-04-29:
@magdesign I'm not at the stage of this project where it's appropriate to test solutions, especially when they're known to not be working. 
I just asked the dev if they happen to know any of the big-picture problems since they said rpi4 is definitely not supported and I've helped the oF rpi4 porting efforts in the past, didn't mean for that to come off as lazy or to upset anyone.

magdesign, 2020-04-29:
It takes a lot of time to test things, if someone can tell: I installed this,  compiled this, that was working but stuck here, its much easier for a dev to fix the issue. That's how opensource works.

So why not installing OF on your RPI4, compile PiMapper and test if it works or what errors you get... the community will help.

eshkrab, 2020-04-29:
Thank you for your input, I'll post if I end up using this and have issues I can't fix myself

sxy2069, 2020-05-07:
> It takes a lot of time to test things, if someone can tell: I installed this, compiled this, that was working but stuck here, its much easier for a dev to fix the issue. That's how opensource [...]

[image]
when I compile example_basic in ofxPimapper addon on Raspberry 4B, I get this error ,who can help me to solve this problem

kr15h, 2020-05-07:
Hi!

Seems like the same issues are there when I last tested. There are some problems with the new gl stuff. I believe a new window class has to be created. @jvcleave seems to be working in that.

cheers

—-
Krisjanis Rijnieks
Creative coding. Projection mapping. Solutions
https://rijnieks.com

jvcleave, 2020-05-07:
unlikely to have texture mode support on the RPI4 
https://github.com/jvcleave/ofxOMXPlayer/issues/157

mysticcircuits, 2020-05-08:
Hey so the odd thing is that I seem to be getting the exact same error on a Pi 3B+.  Did a fresh install of the latest stable version of Noobs and open frameworks today to make sure everything was installed correctly and I got the same error.  Any idea what I might have done wrong?  I had no issue getting ofxOMXPlayer working in the past.

jvcleave, 2020-05-08:
I don't have an Pi 3B+ to test but it looks like maybe `eglCreateImageKHR` has been moved in Buster

If that is the case, there was a new way I was using to access it on the RPI4 that may work. See this commit

https://github.com/jvcleave/ofxOMXPlayer/commit/94178e69d2be5e7ba45c724e3c666009896a7ba1

the important parts are in this block
```
static PFNEGLCREATEIMAGEKHRPROC createImageProc = NULL;
EGLImageKHR createImage(EGLDisplay dpy, EGLContext ctx, EGLenum target,
                        EGLClientBuffer buffer, const EGLint *attrib_list)
{
    if (!createImageProc)
    {
        createImageProc = (PFNEGLCREATEIMAGEKHRPROC)eglGetProcAddress("eglCreateImageKHR");
    }
    
    return createImageProc(dpy, ctx, target, buffer, attrib_list);
}
static PFNEGLDESTROYIMAGEKHRPROC destroyImageProc = NULL;

EGLBoolean destroyImage(EGLDisplay dpy, EGLImageKHR image)
{
    if (!destroyImageProc)
    {
        destroyImageProc = (PFNEGLDESTROYIMAGEKHRPROC)eglGetProcAddress("eglDestroyImageKHR");
    }
    
    return destroyImageProc(dpy, image);
}

bool ofxOMXPlayerEngine::generateEGLImage()
{
    bool success = false;
    bool needsRegeneration = false;
    
    if (!texture.isAllocated())
    {
        needsRegeneration = true;
    }
    else
    {
        if (texture.getWidth() != videoWidth && texture.getHeight() != videoHeight)
        {
            needsRegeneration = true;
        }
    }
    
    if (!fbo.isAllocated())
    {
        needsRegeneration = true;
    }
    else
    {
        if (fbo.getWidth() != videoWidth && fbo.getHeight() != videoHeight)
        {
            needsRegeneration = true;
        }
    }
    
    if(!needsRegeneration)
    {
        //ofLogVerbose(__func__) << "NO CHANGES NEEDED - RETURNING EARLY";
        return true;
    }
    
    if (appEGLWindow == NULL)
    {
        appEGLWindow = (ofAppEGLWindow *) ofGetWindowPtr();
    }
    
    if (appEGLWindow == NULL)
    {
        ofLogError(__func__) << "appEGLWindow is NULL - RETURNING";
        return false;
    }
    if (display == NULL)
    {
        display = appEGLWindow->getEglDisplay();
    }
    if (context == NULL)
    {
        context = appEGLWindow->getEglContext();
    }
    
    if (display == NULL)
    {
        ofLogError(__func__) << "display is NULL - RETURNING";
        return false;
    }
    if (context == NULL)
    {
        ofLogError(__func__) << "context is NULL - RETURNING";
        return false;
    }
    
    if (needsRegeneration)
    {
        
        fbo.allocate(videoWidth, videoHeight, GL_RGBA);
        texture.allocate(videoWidth, videoHeight, GL_RGBA);
        texture.setTextureWrap(GL_REPEAT, GL_REPEAT);
        textureID = texture.getTextureData().textureID;
    }
    
    ofLog() << "textureID: " << textureID;
    ofLog() << "tex.isAllocated(): " << texture.isAllocated();
    ofLog() << "videoWidth: " << videoWidth;
    ofLog() << "videoHeight: " << videoHeight;
    ofLog() << "pixels: " << videoHeight;
    
    // setup first texture
    int dataSize = videoWidth * videoHeight * 4;
    
    if (pixels && needsRegeneration)
    {
        delete[] pixels;
        pixels = NULL;
    }
    
    if (pixels == NULL)
    {
        pixels = new unsigned char[dataSize];
    }
    ofLog() << "dataSize: " << dataSize;
    
    //memset(pixels, 0xff, dataSize);  // white texture, opaque
    
    glBindTexture(GL_TEXTURE_2D, textureID);
    glTexImage2D(GL_TEXTURE_2D, 0, GL_RGBA, videoWidth, videoHeight, 0,
                 GL_RGBA, GL_UNSIGNED_BYTE, pixels);
    
    
    if (eglImage && needsRegeneration)
    {
        destroyEGLImage();
    }
    
    // Create EGL Image
    eglImage = createImage(display, context, EGL_GL_TEXTURE_2D_KHR, (EGLClientBuffer)textureID, NULL);

    if (eglImage == EGL_NO_IMAGE_KHR)
    {
        ofLog()    << "Create EGLImage FAIL <---------------- :(";
        
    }
    else
    {
        success = true;
        ofLog()  << "Create EGLImage PASS <---------------- :)";
    }
    return success;
    
}

void ofxOMXPlayerEngine::destroyEGLImage()
{
    
    
    if (eglImage)
    {
        if (appEGLWindow == NULL)
        {
            appEGLWindow = (ofAppEGLWindow *) ofGetWindowPtr();
        }
        
        if (display == NULL)
        {
            display = appEGLWindow->getEglDisplay();
        }
        
        if (!destroyImage(display, eglImage))
        {
            ofLog() << __func__ << " FAIL <---------------- :(";
        }else
        {
            ofLog() << __func__ << " PASS <---------------- :)";
        }
        eglImage = NULL;
    }
    
}
```

pietrondo, 2020-05-29:
news about pi 4?
thx

kr15h, 2020-05-29:
Nope. I have the feeling that one has to make a new Pi Mapper for Pi 4 actually.

Brumeux, 2021-09-01:
Anything new @kr15h ? Thanks a lot

ali-layken, 2025-02-22:
Hi fellow Pi 4 B users,

everything seems to be working fine using the [of v0.12.0 linuxaarch64 build](https://github.com/openframeworks/openFrameworks/releases/download/0.12.0/of_v0.12.0_linuxaarch64_release.tar.gz) 

I used `addons.norpi.make` to build the examples. Videos were working fine without OMXPlayer. I haven't tested audio but everything seems fine so far:

[image]

[image]

kr15h, 2025-02-22:
No way. You did it!

simonklotz, 2025-03-02:
@ali-layken awesome! And such a perfect timing! I just stumbled upon this project this week.
It would indeed be very helpful if could explain what exactly you did to make it run.

I am not a C++ dev and I don't know what it means if you say:
> I used addons.norpi.make to build the examples.

I found an "addons.make.norpi" file in the example_basic dir. But don't know what I should do with it.

Thanks!

simonklotz, 2025-03-02:
Thanks @ali-layken! That information came perfectly in time, as I just stumbled upon ofxPiMapper this week. :)

I got it running on a Pi 5!

Here is exactly what I did:
https://github.com/simonklotz/pi-mapper/blob/main/README.md

ali-layken, 2025-03-02:
> That's fantastic, could you provide some more information on how you did it?
> 
> edit: I also managed to get it to work, i didn't use the `startx` command to run the x server on headless.

Sure, I renamed `addons.norpi.make` to `addons.make` and make sure to use a `aarch64` ofx build. For everything else I followed ofx's Linux [install instructions](https://openframeworks.cc/setup/linux-install/) & the README of this repo. OMXPlayer is not needed on the Pi 4 B.

ali-layken, 2025-03-02:
> Thanks [@ali-layken](https://github.com/ali-layken)! That information came perfectly in time, as I just stumbled upon ofxPiMapper this week. :)
> 
> I got it running on a Pi 5!
> 
> Here is exactly what I did: https://github.com/simonklotz/pi-mapper/blob/main/README.md

@simonklotz Amazing docs! Nice stuff. Don't forget to use `make -j$(nproc)` as it saves a ton of time when developing.

simonklotz, 2025-03-02:
@ali-layken Thanks! I will keep that in my mind!
I'm a frontend dev and have no idea about openFrameworks and C++ haha. I had the idea of making some visuals for a music location and found this beauty of an open source project.
At first I just wanted to find out if this actually works, and now my next steps are understanding this stuff and learning a little C++ if needed.

ali-layken, 2025-03-02:
> [@ali-layken](https://github.com/ali-layken) Thanks! I will keep that in my mind! I'm a frontend dev and have no idea about openFrameworks and C++ haha. I had the idea of making some visuals for a [...]

Hey, I am learning too I'm full stack but I have never written C++

Check out this fork I am working on to support using other windows as a source: [X Source](https://github.com/kr15h/ofxPiMapper/pull/191) if you want we can connect on discord @supaboop or anywhere and maybe we can both learn some C++ and low level graphics libs.

Also since we are reporting so much success on Pi 4 B, lets wait for @kr15h to check it out and then update this repo's README and close this issue.

simonklotz, 2025-03-02:
Sounds great! I will hit you up on discord!

MoorPoor, 2025-06-19:
somehow it is also working here:
Raspi 4 

Install: 
- could not use latest OpenFrameworks. it compiled fine, but piMapper gave problems.
so i used the one @ali-layken [mentioned above ](https://github.com/openframeworks/openFrameworks/releases/download/0.12.0/of_v0.12.0_linuxaarch64_release.tar.gz)

- I erased the omx player from addons.make as mentioned above.

- changed the main.cpp to go FULLSCREEN

Todo/Problems:
- in "4" view the checkboxes for videos are there but no titels.
- no extra features like automatic loading of videos from USB stick and system shortcuts like "sdn" (shutdown).... might be because Im̀ running in Desktop?
- for now I could not get 2 monitors working. That would be the one big benefit of using a raspi 4 / 5

[image]