kr15h/ofxPiMapper #171: Edge Blending [new feature]

magdesign, 2020-02-21:
We want to add an edge blending function to ofxPiMapper.

It should be draggable from the sides of the quads and create a gradient to black.
Any hints on how to solve this are very welcome (since I am a oxf beginner and do not know every module yet).

kr15h, 2020-02-21:
Hi! Yes this sounds great! I'm just in the middle of teaching and troubles with time, but that is a stupid excuse. Will get myself a bit more organised soon.

magdesign, 2020-02-29:
I made a readme with the wanted features:
https://github.com/magdesign/ofxPiMapper_new_features/blob/master/README.md

My skills in Openframeworks are just too bad to code it myself. If anyone is able to solve one of these features, contact me.

pierrep, 2020-03-12:
@magdesign  I might be able to help with some of those features if you're happy to support my development time.

magdesign, 2020-03-13:
@pierrep: this sounds great :-) , contacted you via form on sensoryempire.com so we can discuss further.
Think first priority would be a working remote app for windows..
 You can also contact me via info_at_magdesign_ch.

pierrep, 2020-06-11:
Hi @kr15h , I've been working with @magdesign on some new features. I just recently sent a pull request for some fullscreen quad code. In another branch in my fork I have an edge blending feature waiting to be merged, but it's a big modification as I had to shift to GLES2 to get shaders to work. It's all functional, apart from some performance regressions in the GLES2 version of ofxGui. The ofDrawCircle() and gui checkboxes seem to be the culprits. Anyway, just wanted to bring this to your attention, would be great to get your feedback.

kr15h, 2020-06-11:
Hi! Thanks for reaching out. I will check and merge as soon as possible.

MoorPoor, 2021-01-20:
Hei together!
I just wanted to bring this up again as an edge blending feature would be a great tool to have.
from my point of view ofxPiMapper is such a great piece of software fitting perfectly in that gap of overblasted mapping works and art installations that I would be more than happy to see any further development process.
Stay healthy!

pierrep, 2021-01-21:
> Hei together!
> I just wanted to bring this up again as an edge blending feature would be a great tool to have.
> from my point of view ofxPiMapper is such a great piece of software fitting perfectly in [...]

Hi,
Edge blending is already implemented in my fork here:
https://github.com/pierrep/ofxPiMapper/

There's a dedicated branch for it, see here:
https://github.com/pierrep/ofxPiMapper/branches

It's also integrated into my master branch. I haven't made a PR because it's a big set of changes, I've had to convert ofxPiMapper to GLES2 to get it working, but if there's interest from @kr15h we could make it happen.

MoorPoor, 2021-02-15:
thanks pierrep for your work, that would be great!
can you explain a bit more, what the use possibilities of the shaders would be you mentioned as benefit of GLES2 to make us even more hungry?!

pierrep, 2021-02-15:
At the moment I've just implemented an edge blend shader into quads. What this means is that you can edge blend multiple projectors that are being driven by ofxPiMapper.

Other than that, with a bit of coding you could also add your own shaders, e.g. using the FBO source and some custom code. This would give you the option of shader-based generative sources. Have a look at https://www.shadertoy.com/ to see the possibilities of shaders. However, the GPU on the RPI is weak, so it won't be able to do too much, but some simple shaders could definitely spice things up. I've used an optimised Perlin noise shader to drive visuals before on an RPI, for example.

MoorPoor, 2021-02-15:
great, a invert shader would be a good thing, specially in combination with a camera source.

MoorPoor, 2021-03-01:
@pierrep would it be possible for you to provide an image of your fork somehow?
As compiling is due to the lack of skills not an option for me.
many greetings

pierrep, 2021-03-02:
@MoorPoor the work I did was for @magdesign  who has builds and images as part of his [pocket VJ](https://www.pocketvj.com) project, you should definitely speak to him about this.