kr15h/ofxPiMapper #172: Compile ofxPiMapper/example_remote-client/ on Windows

magdesign, 2020-03-02:
Can someone compile (or at least tell me how to compile) the ofxPiMapper/example_remote-client/
on a Windows computer?

As a Linux user I tried in a VM with  MSYS2, but stuck in 
`config.addons.mk:210: *** missing separator. `
error.
After spending hours on google I ask here for help, maybe someone has it already compiled and as a working exe.

(Remember to use the `addons.make.norpi ` and include all used dlls with 
'make copy_dlls')

kr15h, 2020-03-02:
oh yes, there is this one block that has to be removed. hold on.

kr15h, 2020-03-02:
ah no that was when compiling of. can you compile a basic openframeworks example on msys2? does it work?

kr15h, 2020-03-02:
sry closed by accident

magdesign, 2020-03-02:
All examples with only on addon in addons.make do compile.
it seems to be a bug on openframeworks.
Also tried with the VisualCode version, but I do not get how it works...

On Linux everything compiles fine and works nice...

magdesign, 2020-03-03:
With VisualStudio2017. 
I first open the Project Generator, the select the ofxPiMapper folder and update the project, then the same with the example_remote-client folder.
When done clicking on: open in IDE, which opens Visualstudio, in there I click Build Solution, which starts a build process and a few minutes later the build failed with 13335 errors... 

tried several attempts with MSYS2, openframeworks and nightly, manually copied all the missing files to the  corresponding folders, but still no luck.

kr15h, 2020-11-16:
Trying out GitHub Actions. Next step - set up compile on Windows.