kr15h/ofxPiMapper #173: Raspberrypi 3B+ & ofxPiMapper & ofxOMXPlayer & openFrameworks

loufoo, 2020-03-15:
Hello @kr15h and @everyone , thanks for ofxPimapper !

I'm in the need to ask your help, trying to make ofxPiMapper on a RPI3B+, with no luck from around 5 days full time (thanks to COVID-19 forced isolation).

> SPOILER: this is boring
> I tried most of the solutions i found online, and multiple raspbian versions:
> - jessie lite image from ofxPimapper website (not working on rpi3b+)
> - a Pocket vj image i made few years ago with control panel 2 based on jessie lite
>   (works on pi3 not 3b+, not able to repair circular dependencies problems with libraries, repos updated )
> - switching a jessie lite from a rpi3 to a rpi3b+ and rpi-update (repos changed, compile of problems, gstreamer problems, cairo problems)
> - manual updating drivers, kernel, boot(partitioning manually) from jessie to stretch ((of version 0.11.0 and 0.10.0 miss 0.10.1,repos changed, compile of problems, gstreamer problems, cairo problems, libs broken)
> - manual updating from stretch to actual stable buster(of version 0.11.0 and 0.10.0 miss 0.10.1)
> - starting from strech raspbian lite(no luck)
> - starting from a buster raspbian lite(even less luck)
> -also tried multiple versions of ofx on most of the distro versions.

As none of my attempts is taking me anywhere, i'm asking to guide me in spinning Pimapper and ofxOMXPlayer & Openframeworks on RpiB3+.

Thank you in advance for your help and this piece of software.
Cheers

kr15h, 2020-03-15:
Oh that is what this covid isolation should work best for if not recovering yourself from it. it should work with Raspberry Pi 3B+. It is not ported to Pi 4 yet. Are you sure your Raspberry Pi or SD card is fine?

kr15h, 2020-03-15:
I think you mentioned, but the image from ofxpimapper website should work https://ofxpimapper.com/
It is also true that the project itself requires a serious rework. I am currently very bad at planning this, but I am not going to give up as I think it important to have a nice and versatile open source projection mapping tool.

loufoo, 2020-03-15:
> Oh that is what this covid isolation should work best for if not recovering yourself from it. it should work with Raspberry Pi 3B+. It is not ported to Pi 4 yet. Are you sure your Raspberry Pi or SD card is fine?

yep, luckly i'm not affected by covid directly, but as i'm in italy i'm home as anyone else atm.

i have 8 rpi at home atm, 6 are rpi3b+, 1 rpi3, 1 rpi A+, 1p0w XD

i'm working on 2 rpi3b+ and a rpip3 just now, all three cards are brand new
and flashed fresh, from official images and old ones that are working on rpi3.

I'll try again the image from ofxPimapper website on a pi while with a fiend we are trying to spin a containerised version on a buster lite. i'll let you know in a few minutes : )

loufoo, 2020-03-15:
[https://pasteboard.co/IZgbmHK.jpg](url)
> I'll try again the image from ofxPimapper website on a pi while with a fiend we are trying to spin a containerised version on a buster lite. i'll let you know in a few minutes : )

i confirm the image you have on the website works with raspberrypi3b+, the problem was probably on my side probably in writing the image.

Now i Wrote it with etcher and it boots and goes in mapper, flawlesly.

What are the specs for these installation?

kr15h, 2020-03-16:
OK. That is good to know. 
The current image is based on `2019-04-08-raspbian-stretch`.
And the folowing are commit IDs for dependencies.
```
OFX_VERSION="68eb8a7333f2d301268a41028ceafdd2f0d7d894"
OFX_JSON_VERSION="55a5825ddf510c991466f9d39d8e1d57b2bc2008"
OFX_OMXPLAYER_VERSION="3c047ed51452d1ab0a3deb60ac57e9dd54bb2938"
OFX_PIMAPPER_VERSION="55b3c165df7cf1fb48356f2bb159a659195548ca"
OFX_OMX_CAMERA_VERSION="7871b9c91a79771076d30c6610c774c9bcd1ac16"
```
For more details please consult the CI configuration here: https://gitlab.com/kriwkrow/pimapper/-/tree/master

closed 2020-11-16 by kr15h