kr15h/ofxPiMapper #176: Remote Client

balejandrovb, 2020-08-31:
Hi thanks 4 this documentation
I Can´t achieve to open in web browser the interface ... i dont know what im missing to set .
i had clone all the things what this forum said .... please help mee
im from mexico sorry 4 my English 
Thanks

magdesign, 2020-09-01:
Its not meant to be to work within the browser.
You need to compile a remote client or grab the pre-compiled library here: https://github.com/magdesign/ofxPiMapper_PVJexh/releases/tag/v0.1

Please consider to make a donation: 
https://www.patreon.com/kriwkrow
https://ofxpimapper.com/

balejandrovb, 2020-09-01:
then the only thing i need to do is execute that .exe?

magdesign, 2020-09-01:
define the ip in config.json

give yourself some time, use google, read the readme files: 
https://github.com/kr15h/ofxPiMapper/tree/master/example_remote-client
https://github.com/kr15h/ofxPiMapper/tree/master/example_remote-server

then write a documentation howto in this issue for the others.

balejandrovb, 2020-09-01:
thanks for the info.
but i read all the files. i´ve define the ip in config. json , i cant execute correctly the app

magdesign, 2020-09-02:
I have no idea on windowsOS, maybe the firewall is blocking it...

danrouch11, 2020-09-03:
How can I run the ./example_remote-client from the terminal?

BjarneJensen, 2021-03-03:
I noticed on windows, I get errors if the sources are not there, so make sure you have  bin/data/sources/images and bin/data/sources/videos directories with the same content as your server. This is what worked for me. Alternatively, you could try building the linux version in a virtual environment if you want to make changes (i could never get the windows version to work in visual studio and am not familiar with MSYS2)