kr15h/ofxPiMapper #177: what is the command line to open ofxpimapper

balejandrovb, 2020-09-09:
my program closed and no longer opens, what is the command line to execute it. I already tried restarting the RPb3 but it doesn't work
... please help me

kr15h, 2020-09-09:
Hi. Sorry to read. What was the last thing you did when it was still open?

balejandrovb, 2020-09-09:
i restarted because i introduced USB for new sources. i dont want to reinstall the .img , because i already have the perfect callibration. is there any way to open by command line? thanks for ur attention

kr15h, 2020-09-09:
So yeah it might be that Pi Mapper has problems playing back some of your videos. Place an empty file with the name `PIMAPPER_DELETE_SOURCES` on the USB drive and reboot. Then double-check your sources to be HD 720p resolution and 30fps or so. Use `baseline` H264 setting when encoding. Best results with this so far.

kr15h, 2020-09-09:
Another thing could be that the settings file got broken and needs to be restored. Anyhow one could check it with `sudo systemctl status pimapper`

closed 2020-11-16 by kr15h