kr15h/ofxPiMapper #178: Problem compiling ofxOMXPlayer: undefined reference to symbol 'eglCreateImageKHR'

Dazzid, 2020-10-05:
I'm trying to run the ofxPiMapper in RaspBerry Pi 3 (debian version 10.6, Linux raspberrypi 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux) with OF version 0.11.0. 

After following the instructions, the compiler is having problems with 'eglCreateImageKHR' compiling the ofxOMXPlayer addon.

The message is the following:
/usr/bin/ld: /home/pi/openFrameworks/addons/obj/linuxarmv6l/Release/ofxOMXPlayer/src/ofxOMXPlayerEngine.o: undefined reference to symbol 'eglCreateImageKHR'
/usr/bin/ld: //opt/vc/lib/libbrcmEGL.so: error adding symbols: DSO missing from command line

It seems the issue is already posted in the ofxOMXPlayer forum. However, there is no a clear solution, have you guys resolved this issue?

bsyzek, 2021-01-06:
I'm having the same problem.

fbarretto, 2021-07-16:
Has anyone solved this? I'm struggling to compile ofxOMXPlayer examples at RPI3 running buster and OF 0.11.0.

fbarretto, 2021-07-17:
Just to let everyone know, these steps worked for me:

- As of 0.11.0 the default window is ofAppGLFWWindow. If you want to use the older ofAppEGLWindow approach, comment out USE_GLFW_WINDOW = 1 in libs/openFrameworksCompiled/project/linuxarmv6l/config.linuxarmv6l.default.mk
- Select "legacy driver" in "raspi-config" advanced options.
- Checkout branch rpi-4 develop (https://github.com/jvcleave/ofxOMXPlayer/tree/rpi4-develop)

Cheers!

Dazzid, 2021-09-15:
Hi fbarretto 
I tested this configuration and it doesn't compile in my RBPi 4 with raspbian-buster and Off_v0.11.
I might be missing something relevant? What example did you test?

kr15h, 2021-09-15:
Hi! Raspberry PI 4 is not yet supported. Stick to Pi 3