kr15h/ofxPiMapper #181: Playback restart at preset change

araucarialab, 2021-03-02:
Hello! 
I know that may sound silly but I'm no programmer so I've been trying to get the videos' playback to restart every time I change the preset. I saw that there is a reset() function but I just can't make it work. I've been trying to do like this:

`void ofApp::keyPressed(int key){

        switch (key) {
                case '5':
                        mapper.reset();
                        mapper.setPreset(0);
                        break;`

But with no luck. 
Sorry if this is a silly question but I've been trying for the last two days and I just can't figure it out.
Thanks in advance!

araucarialab, 2021-03-07:
I also noted that with 2 presets I can barely work with two videos, is it normal? How can I address this? Because without the presets I can run like 10 videos. 
And again, sorry for the entry level programming here, I just don't know where I can find more info about this to try to solve by myself, I really tried for some days after coming here to ask for help. 
Thanks!

magdesign, 2021-03-07:
What I do to change presets is to kill ofxPiMapper then copy the new .xml file to 
.../ofxPiMapper/example/bin/data/ofxpimapper.xml

all with a simple bash script:
see here: https://github.com/magdesign/PocketVJ-CP-v3/blob/master/sync/mapperimport1

kr15h, 2021-03-08:
That is indeed a nasty way of doing it @magdesign . The core of the problem is the video player which is still based on ofxomxplayer. One should revisit that. When changing presets now, the videos are not being unloaded from memory and it is generally the problem with the current video player implementation.

araucarialab, 2021-06-15:
Thanks guys! After some tries I finally got it. Actually I needed something random but I did this part using an Arduino as a keyboard. The final code I used was based on @magdesign 's one:

`/home/pi/ofx/addons/ofxPiMapper/code01/./bin/code01 -f &

sleep 3

sudo killall -9 /home/pi/ofx/addons/ofxPiMapper/code02/./bin/code02 > /dev/null 2>&1

sleep 0.5`

I will organize everything to have it on my github too!

closed 2021-06-15 by araucarialab