kr15h/ofxPiMapper #183: A flask frontend for ofxPimapper

l00f00, 2021-06-29:
Hello this is a funny frontend i've made during the pandemic for controlling pimapper with my phone https://github.com/l00f00/flasketvj
It's made with python/flask and controls the mapper directly sending keystrokes and commands, inspired by PocketVj from MagDesign and totally build around ofxPimapper and the shell.

Hope it usefull. Cheers! And thanks again @kr15h for this software, i love it!

kr15h, 2021-06-29:
Wow thanks for this @l00f00 ! I could try to integrate this in the next rpi image release.

l00f00, 2021-06-29:
Yeah, great! Let me know if i can aid you in the process.

fbarretto, 2021-07-13:
Great work! Btw, when will a newer pimapping rpi image be available? I've never managed to run v.1.2.0 in a RPI4. 

Cheers!