kr15h/ofxPiMapper #184: Try to adjust volume but still seem not much louder

yao-bite, 2021-07-25:
Hello, I'm new here. 
First thanks for the amazing mapping tool!

My problem is, after trying my video
audio seems to work pretty good, but in very small volume.
At first I thought the problem was the audio track itself, 
but then after trying to play only one video (with same volume audio output) with other code ( which uses omxplayer)
I found that while using ofxPiMapper, the audio volume is much smaller.

I also have tried to adjust the volume with these steps:
"Open alsamixer.
alsamixer
Set the volume to a value between 90 to 100 by using the arrow keys. ESC to exit the mixer."
But seems no big difference between 50 and 100.

Was there something that I missed?

kr15h, 2021-08-31:
Hi. Sorry, I have not bumped into this issue myself. What I could assume is that omxplayer has its own way of setting the volume and it could be that one should be able to define it via code.

magdesign, 2021-08-31:
Before compiling change the volume settings in:
`
ofxOMXPlayer/src/ofxOMXPlayerSettings.h => initialVolume from 0.3 to 0.5`

Then you have 100% volume.

magdesign, 2022-06-23:
Does anyone know how to adjust the volume when mapper is playing over 3.5 mm jack (bcm2835) ?
Since adjusting the volume in alsamixer does not make any changes.

It seems to be implemented in ofxOMXplayer, so we need to send some kind of dbus command, but how?
or any other nice hacks/workarounds?

https://github.com/jvcleave/ofxOMXPlayer/issues/21