kr15h/ofxPiMapper #187: Simple Grid Surface

magdesign, 2022-01-23:
Anyone know how to add a grid surface to ofxPiMapper?

Meaning a surface which is split into 4 parts as shown here:
[image]

So we are able to move the center point and also the horizontal and vertical middle.
I know that it should be solvable with a mathematical theorem (german=> Strahlensatz) solution.

Currently when I use several quads in ofxPiMapper it will not fit:
[image]

What I found is this paper: https://www.ronenbarzel.org/papers/warp.pdf
See on page 13-14
and also this paper:
https://www.inf.ufrgs.br/~oliveira/pubs_files/PG01_Adaptive_subdivision.pdf

Maybe someone good in mathematics can tell us the formula to code this?

kr15h, 2022-01-23:
There is a grid surface. Have you tried hitting g in mapping mode?

kr15h, 2022-01-23:
True it does not have perspective yet.

magdesign, 2022-01-24:
> True it does not have perspective yet

I tried on other mapping tools too, stitching quad surfaces together does nowhere work, so there is the need of quad-grid surface.
Maybe @pierrep (https://github.com/pierrep/ofxPiMapper) can help us on this. Any inputs from people familiar with geometric calculations, mathematics, coding very welcome...