kr15h/ofxPiMapper #189: Trouble opening basic example on Rpi 3B on different OF versions

owenplanchart, 2022-11-17:
Hi, 
Thank you for building this tool.

I am trying to use this addon on a Rpi 3B. 

I started out using OF 11.0 and debian buster kept having problems with the cairo library. I downloaded cairolib-dev but there were still a lot of linker errors.

I then went back to this page and just downloaded the disk image: https://ofxpimapper.com/. Which I noticed used Jessie and a much older version of OF. It worked perfectly out of the box.

I then downloaded OF 10.1 onto  a fresh install of Jessie and had similar errors.

Can you help me understand which version of raspbian and which version of OF would work best to get the examplesFboSources  running on the Rpi3b?

owenplanchart, 2022-11-17:
HOST_OS=Linux
checking pkg-config libraries:   cairo zlib gstreamer-app-0.10 gstreamer-0.10 gstreamer-video-0.10 gstreamer-base-0.10 libudev freetype2 fontconfig sndfile openal libcurl 
with PKG_CONFIG_LIBDIR=
Package cairo was not found in the pkg-config search path.
Perhaps you should add the directory containing `cairo.pc'
to the PKG_CONFIG_PATH environment variable
No package 'cairo' found
Package cairo was not found in the pkg-config search path.
Perhaps you should add the directory containing `cairo.pc'
to the PKG_CONFIG_PATH environment variable
No package 'cairo' found
/home/pi/openFrameworks/libs/openFrameworksCompiled/project/makefileCommon/config.shared.mk:291: *** couldn't find cairo pkg-config package or it's dependencies, did you run the latest install_dependencies.sh?.  Stop.

kr15h, 2022-11-18:
Hi. It could be that some of the libraries needed for openFrameworks did no install. Did you run the `scripts/linux/download_libs.sh` script?

owenplanchart, 2022-11-19:
thanks for the prompt reply. Very new to git and linux: am I doing something wrong here:
pi@raspberrypi:~/openFrameworks $ scripts/linux/download_libs.sh
bash: scripts/linux/download_libs.sh: No such file or directory
pi@raspberrypi:~/openFrameworks $ cd
pi@raspberrypi:~ $ scripts/linux/download_libs.sh
bash: scripts/linux/download_libs.sh: No such file or directory
pi@raspberrypi:~ $ 

Inside the scripts/linux folder I can definitely see other folders with all the different linux systems and other .h and .sh files 

Thanks