kr15h/ofxPiMapper #190: Resources GUI: no file names, only checkboxes in OF 11, windows

dasoe, 2023-06-07:
Hi all,

as always **_cheers(!)_** for this incredible piece of software!

when building _exaple_basic_ with OF 11.2, on Windows, I have only checkboxes in the Resources GUI, no names of files.
when building _exaple_basic_ with OF 10.1, same machine, it looks as expected. (checkboxes and names of files).

It seems, there's some bug or breaking change in OF 11 ofxGui? I don't know if there's other parts of the code, that might need adjustment, too, nore if there's a better solution, but as a quick fix, I used the following change to solve the issue for me:

```
// ofxPiMapper/src/UserInterface/Radiolist.cpp, Line 45:

void RadioList::setup(std::string title, std::vector<std::string> & labels, std::vector<std::string> & values){

	// Store title for later use
	storedTitle = title;
	guiGroup.setName(title);
+	guiGroup.setWidthElements(220.0f);
	setup(labels, values);
}
```

I used a hardcoded width value, as the position is hardcoded too, in `ofxPiMapper/src/Gui/Widgets/SourcesEditorWidget.cpp`

have a great day!
oe

kr15h, 2023-06-13:
I will need to check this, but it is not going to be super fast.

doctea, 2023-09-28:
I'm getting the same problem with OF 12.0 on Linux.  Your workaround does the job for me, thanks @dasoe !